vitotai/BrewPiLess #436: Sonoff TH Origin

jonathanschneider, 2024-10-19:
Hello,
I am trying to use BrewPiLess on a Sonoff TH Origin which has an ESP32 built in. After flashing the device and restarting I can't find the AP. I've tried flashing the binary [BrewPiLess.v4.4.esp32.bin](https://github.com/vitotai/BrewPiLess/releases/download/4.4/BrewPiLess.v4.4.esp32.bin) with NodeMCU PyFlasher as well as uploading `esp32-dev` and `esp32` through PlatformIO.
Any ideas? Am I missing something?
Thanks!

zandegran, 2024-10-19:
I had the same problem with Sonoff TH Elite and now i am able to flash via VSCode and now i am having another problem..

vitotai, 2024-10-21:
You will need to figure out the configuration of the board to make it work, including
- onboard Flash size
- GPIO PINs used
- other parameters that I might not know.

jonathanschneider, 2024-10-21:
The flash size is 4MB, so `board = esp32dev` should be right. I can figure out the PINs. Would the wrong PINs keep the device from booting though?

jonathanschneider, 2024-10-22:
I figured out my problem. Flashing one of the ESP32 envs the ESP was stuck searching for an LCD. The Sonoff TH Origin doesn't have one. I combined the existing envs for ESP32 and the old Sonoff devices and now got it to boot. I will create a PR soon.

jonathanschneider, 2024-10-22:
To create the PR I also need to adjust Config.h. Since the Sonoff TH Origin uses an ESP32 I want to make sure all its options are selected. Where is this declared?

https://github.com/vitotai/BrewPiLess/blob/710639aa7dfaea9fae58f676ea3ded8d64c25ca9/src/Config.h#L316

zandegran, 2024-10-23:
Here is the PR that should work for both Origin and Elite including Leds https://github.com/vitotai/BrewPiLess/pull/438

@jonathanschneider @vitotai