vitotai/BrewPiLess #437: Sonoff TH 16 Elite cannot detect temp sensor

zandegran, 2024-10-19:
The temp prob data [should be at GPIO25](https://templates.blakadder.com/sonoff_THR316D.html). I changed the config to reflect this. Changed `oneWirePin` from 23 to 25 in `Config.h`. GPIO27 gives power to the sensor so added the following to `void brewpi_setup()` in `BrewPiLess.cpp`

```
void brewpi_setup()
{
  pinMode(27, OUTPUT); 
  digitalWrite(27, LOW);
  pinMode(13, OUTPUT); 
  digitalWrite(13, LOW); // Green status Led
  ....
}
```

The LED is turning on but the temperature sensor is not detected.

vitotai, 2024-10-21:
I don't think GPIO would give enough power to OneWire Device. According to my limited experience, even 3.3V from ESP32 didn't work well. I am NO hardware expert, though.

lalo-uy, 2024-10-21:
18B20 works well from 3 to 5v. It depends on the wire length. No problem
with 2m and a 3k3 pullup. They stuck from time to time, so is a good option
power from gpio and be aple to remote reset the sensor.

zandegran, 2024-10-21:
This is standard 1.5m cable that came with the sonoff th16 elite device. i didn't check if it worked with the original firmware.. Will do it later today.. @lalo-uy the sonoff sensor has pullup built in. 

I updated the one-wire lib to 2.3.8 and added the line `#define ARDUINO_ARCH_ESP32 true` to `config.h `
I guess this should fix some timing issue with some of the sensors. 
**[https://github.com/PaulStoffregen/OneWire/blob/master/OneWire.cpp#L151](https://github.com/PaulStoffregen/OneWire/blob/master/OneWire.cpp#L151)**

> I don't think GPIO would give enough power to OneWire Device. According to my limited experience, even 3.3V from ESP32 didn't work well. I am NO hardware expert, though.

jonathanschneider, 2024-10-22:
I have a Sonoff TH Origin that I'm trying to get to work at the moment. The sensor I received isn't just a DS18B20. There is another microcontroller installed in the cable and I believe from there on it's not OneWire anymore. Does your temp sensor cable also have a little box in it?

zandegran, 2024-10-22:
Yes it does... I was already thinking of buying a RJ11 to TRS adapter that Sonoff sells so that I can test with sensor that I know works.

zandegran, 2024-10-22:
Bingo! I spliced the old device and everything is working... Big thanks @jonathanschneider 
I have a [version](https://github.com/zandegran/BrewPiLess/tree/wip-sonoff-th-16-elite) where 
Red led indicates power
Blue led indicates wifi connection
Green led indicates cooler/heater(relay) on

I will receive oled-lcd tomorrow... I will create a PR once I get that working

zandegran, 2024-10-22:
So, technically the [Sonoff AL010 2.5mm Audio Jack to RJ9 Adapter](https://itead.cc/product/al010-2-5mm-audio-jack-to-rj11-adapter/ref/34/) could work with old sensor which are still easy to find

closed 2024-10-22 by zandegran

zandegran, 2024-10-23:
[image]

zandegran, 2024-10-23:
@jonathanschneider Here is the PR https://github.com/vitotai/BrewPiLess/pull/438

jonathanschneider, 2024-10-27:
@zandegran everything works fine except that I can't get the device to cool. I've setup the "devices" but it won't go from idle to cooling and thus not activate the relay (or the associated LED). Both the TH Elite and TH Origin should have the same pinout. Did you do anything special in setting up the devices?

jonathanschneider, 2024-10-28:
@zandegran I can confirm that the PR works for TH Origin as well. My issue is that I can't use Beer Profile or Beer Constant with only one sensor. It seems that BrewPiLess requires both beer temp and fridge temp for the modes to work. My other device is running V4.2 and there only one sensor is necessary.

zandegran, 2024-10-28:
@jonathanschneider I should have checked.. Only the Relays are different not the leds, data and sensorpower though.. It takes some minutes before the cooling begins so the LED might be working already..

jonathanschneider, 2024-10-28:
I had checked for differences between Elite and Origin but not between 16 and 20. The LED does work!

zandegran, 2024-10-28:
@jonathanschneider This PR should fix your problem https://github.com/vitotai/BrewPiLess/pull/440