vitotai/BrewPiLess #439: Can't use Beer Profile or Beer Constant with only one temperature sensor

jonathanschneider, 2024-10-28:
I am running V4.4 on an ESP32 device and have one temperature sensor connected to it. I set up the temperature sensor as "Beer Temp" but BrewPiLess will remain idle indefinitely. If I set up the temperature sensor as "Fridge Temp" and use "Fridge Constant", it will start cooling as expected but I can't get it to cool in the other modes.

I assume that in the other modes both a beer temp and a fridge temp have to be present. Is that right?

I've had V4.2 running on another device for quite some time with only one sensor connected and set up as "Beer Temp". In this case "Fridge Temp" is fed by "Beer Temp". Was that behaviour changed with more recent releases?

How do I use the other modes with only one temperature sensor?

jonathanschneider, 2024-10-31:
@zandegran since you use the RJ9 adapter, you must have only one sensor connected. Did you manage to get Beer profile or Beer constant to work?

zandegran, 2024-12-02:
@jonathanschneider Sorry for the delayed reply. I use iSpindel(external sensor) for beer sensor.. So this does not affect me.