vitotai/BrewPiLess #442: Building v4.4

Rooo6, 2024-11-03:
Trying to build the project (v4.4) i'm getting error messages:
Version 4.3 worked for me and could be installed on esp32-dev

`Processing esp32-dev (platform: espressif32@~6.8.1; board: esp32dev; framework: arduino)
----------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via -v, --verbose option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.8.1) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20017.0 (2.0.17)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 46 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 6.19.3
|-- ESP8266_SSD1306 @ 4.0.0
|-- NimBLE-Arduino @ 1.4.2
|-- AsyncTCP @ 1.1.1+sha.ca8ac5f
|-- ESP Async WebServer @ 1.2.3+sha.43381ab
|-- FS @ 2.0.0
|-- LittleFS @ 2.0.0
|-- SPIFFS @ 2.0.0
|-- WiFi @ 2.0.0
|-- ArduinoOTA @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- OneWire @ 2.3.7
|-- HTTPClient @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
|-- Wire @ 2.0.0
|-- WebServer @ 2.0.0
|-- PolynomialRegression
|-- DNSServer @ 2.0.0
|-- Adafruit ADS1X15 @ 1.0.1
|-- AsyncMqttClient @ 0.8.2
|-- Adafruit BME280 Library @ 2.1.2
|-- Adafruit Unified Sensor @ 1.1.4
|-- Update @ 2.0.0
|-- DHT sensor library @ 1.4.0
|-- EEPROM @ 2.0.0
Building in release mode
Compiling .pio\build\esp32-dev\src\Actuator.cpp.o
Compiling .pio\build\esp32-dev\src\ActuatorArduinoPin.cpp.o
Compiling .pio\build\esp32-dev\src\ArduinoFunctions.cpp.o
Compiling .pio\build\esp32-dev\src\AutoCapControl.cpp.o
Compiling .pio\build\esp32-dev\src\BPLSettings.cpp.o
Compiling .pio\build\esp32-dev\src\BleListener.cpp.o
Compiling .pio\build\esp32-dev\src\BlePillListener.cpp.o
Compiling .pio\build\esp32-dev\src\BleSensorListener.cpp.o
Compiling .pio\build\esp32-dev\src\BleTiltListener.cpp.o
Compiling .pio\build\esp32-dev\src\BrewKeeper.cpp.o
Compiling .pio\build\esp32-dev\src\BrewLogger.cpp.o
Compiling .pio\build\esp32-dev\src\BrewPiLess.cpp.o
Compiling .pio\build\esp32-dev\src\BrewPiProxy.cpp.o
Compiling .pio\build\esp32-dev\src\Brewpi.cpp.o
Compiling .pio\build\esp32-dev\src\BrewpiStrings.cpp.o
Compiling .pio\build\esp32-dev\src\Buzzer.cpp.o
Compiling .pio\build\esp32-dev\src\DS2413.cpp.o
Compiling .pio\build\esp32-dev\src\DallasTemperature.cpp.o
Compiling .pio\build\esp32-dev\src\DataLogger.cpp.o
Compiling .pio\build\esp32-dev\src\DeviceManager.cpp.o
Compiling .pio\build\esp32-dev\src\Display.cpp.o
Compiling .pio\build\esp32-dev\src\DisplayIspindel.cpp.o
Compiling .pio\build\esp32-dev\src\DisplayLcd.cpp.o
Compiling .pio\build\esp32-dev\src\ESP32HTTPUpdateServer.cpp.o
Compiling .pio\build\esp32-dev\src\ESPUpdateServer.cpp.o
Compiling .pio\build\esp32-dev\src\EepromManager.cpp.o
src/BrewPiLess.cpp: In function 'void brewpi_setup()':
src/BrewPiLess.cpp:1706:11: error: 'sensorPowerPin' was not declared in this scope
   pinMode(sensorPowerPin, OUTPUT);  // Power for sonoff temp sensor
           ^~~~~~~~~~~~~~
src/BrewPiLess.cpp:1708:11: error: 'powerIndicatorPin' was not declared in this scope
   pinMode(powerIndicatorPin, OUTPUT);
           ^~~~~~~~~~~~~~~~~
*** [.pio\build\esp32-dev\src\BrewPiLess.cpp.o] Error 1
========================================= [FAILED] Took 15.35 seconds =========================================

Environment    Status    Duration
-------------  --------  ------------
esp32-dev      FAILED    00:00:15.350
==================================== 1 failed, 0 succeeded in 00:00:15.350 ==================================== 

 *  The terminal process "C:\Users\rooo6\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload', '--environment', 'esp32-dev'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. `

ce-designs, 2024-11-11:
I can confirm that building the current master branch is not possible. I get the exact same errors.

roglioburigo, 2024-11-20:
Find all code with if (SONOFF_NEWGEN) blocks and change to #if SONOFF_NEWGEN, idk if it is the best way to solve, but I think this will solve both problems.

dumpamob, 2024-12-04:
other errors on ESP8266

....
src\BrewPiLess.cpp: In member function 'virtual void BrewPiWebHandler::handleRequest(AsyncWebServerRequest*)':
src\BrewPiLess.cpp:540:12: error: 'class ESP8266WiFiClass' has no member named 'eraseAP'
  540 |       WiFi.eraseAP();
      |            ^~~~~~~
src\BrewPiLess.cpp:554:12: error: 'class ESP8266WiFiClass' has no member named 'eraseAP'
  554 |       WiFi.eraseAP(); // https://github.com/espressif/arduino-esp32/issues/8976
      |            ^~~~~~~
src\BrewPiLess.cpp: In function 'void brewpi_setup()':
src\BrewPiLess.cpp:1706:11: error: 'sensorPowerPin' was not declared in this scope
 1706 |   pinMode(sensorPowerPin, OUTPUT);  // Power for sonoff temp sensor
      |           ^~~~~~~~~~~~~~
src\BrewPiLess.cpp:1708:11: error: 'powerIndicatorPin' was not declared in this scope
 1708 |   pinMode(powerIndicatorPin, OUTPUT);
      |           ^~~~~~~~~~~~~~~~~
Compiling .pio\build\esp8266-oled\src\DisplayLcd.cpp.o
src\DS2413.cpp: In member function 'byte DS2413::accessRead(uint8_t)':
src\DS2413.cpp:50:25: warning: suggest parentheses around operand of '!' or change '&' to '&&' or '!' to '~' [-Wparentheses]
   50 |   success = (data>>4)==(!data&0xF);
      |                         ^~~~~
Compiling .pio\build\esp8266-oled\src\ESP32HTTPUpdateServer.cpp.o
Compiling .pio\build\esp8266-oled\src\ESPUpdateServer.cpp.o
*** [.pio\build\esp8266-oled\src\BrewPiLess.cpp.o] Error 1
src\DeviceManager.cpp: In static member function 'static void* DeviceManager::createDevice(DeviceConfig&, DeviceType)':
src\DeviceManager.cpp:137:9: warning: enumeration value 'DEVICE_HARDWARE_BME280' not handled in switch [-Wswitch]
  137 |  switch (config.deviceHardware) {
      |         ^
src\DeviceManager.cpp:137:9: warning: enumeration value 'DEVICE_HARDWARE_BTHOME_HUMIDITY' not handled in switch [-Wswitch]
src\DeviceManager.cpp:137:9: warning: enumeration value 'DEVICE_HARDWARE_BTHOME_THERMOMETER' not handled in switch [-Wswitch]