vitotai/BrewPiLess #445: ESP32 OLED Does not Work

lfach, 2024-11-13:
I´m using Brewpiless-32 PCB and I cant make OLED to work at all.
I tried to add "Wire.begin(22,23);" on "void setup(void){" as someone here suggested, but still does not work.

Im using ESP32-WROM-32D with Vscode + PlatformIO to compile and upload code to the board. 

I tried with ESP8266 and works like a charm. 

Is there anything else that I forgot?

stefschin, 2024-11-13:
Hi, as you can see on the [schematic](https://github.com/stefschin/BrewPiLess-32-PCB/blob/master/BPL32%20Schematic%20Rev%2000.pdf) of the board I2C SDA is connected to GPIO 21 and SCL to 22, so it will never work. Anyway you have to be sure how to use the oled display with the firmware, cheers

lfach, 2024-11-13:
Awesome !!! I was messing up the port numbers....now it worked !!! 
Thanks a lot !!!

lfach, 2024-11-13:
Is there any way to use bigger TFT screen or even a simple 1.3 pol OLED ?

stefschin, 2024-11-13:
Sure it's a way to use a bigger OLED but you have to edit the code to make it work, because 1.3 uses another driver.
For TFT you have to change to SPI communication, change driver and design the screens to show and how they behave.

stefschin, 2024-11-13:
Yes, many but one of the most used I think, it's from [Adafruit](https://learn.adafruit.com/adafruit-gfx-graphics-library/overview)

lfach, 2024-11-19:
Solution provides worked.

closed 2024-11-19 by lfach