vitotai/BrewPiLess #446: Issue to build v4.4

lfach, 2024-11-19:
Hi, 

When I try to build v4.4 on Platform.io in VSCode, I see this error?

src/BrewPiLess.cpp: In function 'void brewpi_setup()':
src/BrewPiLess.cpp:1706:11: error: 'sensorPowerPin' was not declared in this scope
   pinMode(sensorPowerPin, OUTPUT);  // Power for sonoff temp sensor
           ^~~~~~~~~~~~~~
src/BrewPiLess.cpp:1708:11: error: 'powerIndicatorPin' was not declared in this scope
   pinMode(powerIndicatorPin, OUTPUT);
           ^~~~~~~~~~~~~~~~~
*** [.pio\build\esp32-dev\src\BrewPiLess.cpp.o] Error 1
============================================================================ [FAILED] Took 15.47 seconds ============================================================================

Environment    Status    Duration
-------------  --------  ------------
esp32-dev      FAILED    00:00:15.473
======================================================================= 1 failed, 0 succeeded in 00:00:15.473 =======================================================================

Is there any library missing ?

roglioburigo, 2024-11-20:
Find all code with if (SONOFF_NEWGEN) blocks and change to #if SONOFF_NEWGEN, idk if it is the best way to solve, but I think this will solve both problems.

roglioburigo, 2024-11-22:
Hello, take a look in my pull request 448 that I mentioned in the previous comment.
the change isn`t put # before the ifs, is using the if to compile certain code blocks using a system variable.
see in this exemple from the PR:
[image]

lfach, 2024-11-29:
Thanks a lot @roglioburigo . It worked like a charm.

closed 2024-11-29 by lfach