vitotai/BrewPiLess #449: Temp sensors not working properly

lfach, 2024-11-29:
Hello,

I'm using the Brewpiless-32 PCB, with 2 temperature sensors. Since I'm using 2 sensors, I'm using a 4k7 resistor.
I notice that in some of the parts I've assembled, only 1 sensor is working, or none of them work... and if I connect the sensors individually, they work. What could be wrong? Maybe the resistor?

Thanks.

lalo-uy, 2024-11-29:
A lower resistor may help. 3k3 is ok at 3.3v , 4k7 is for 5v.
You can go as low as 1k.
The length of the wires also matters.
Is better to dasy chain the sensors than put too different length joined at
the board.

lfach, 2024-11-29:
Hi @lalo-uy 
I exchanged the resistor to a 2k2 and now I have both sensors been identified (with their hardware addresses) but only one is showing a temp value.
Do you think this is a sensor issue or still resistor that could go lower ? 

Thanks

WojtekGD, 2025-02-16:
Hi
There are many DS18B20 fakes. I had a few of them bought on Aliexpress. They worked with Arduino but didn't work properly with ESP or other devices. When connecting several Dallass they randomly reported errors and I had to choose a pull-up resistor of about 1k or less.