vitotai/BrewPiLess #454: ESP32: Buttons and rotary doesn't work

sigma415, 2025-04-15:
I flashed precompiled  "BrewPiLess.v4.4.esp32.oled.bin" to 38pin ESP32 but neither up/down buttons nor rotary encoder will work.

Up/down buttons are connected to GPIO32 and 33 and both switch to GND.
Alternatively I connected rotary encoder pins L/R to GPIO32 and 33 and push contact to GPIO27. All three of them also switch to GND.
When I measure GPIO27 with scope it's state is always LOW.

What I'm doing wrong ?
Which UI button strategy is implemented in "BrewPiLess.v4.4.esp32.oled.bin" ?

Thanks a lot for help.

vitotai, 2025-04-15:
up/down buttons are not supported for ESP32.  Rotary Encoder is enabled by default for ESP32. 
The pin for Push is 25 by default.

sigma415, 2025-04-15:
@vitotai : Thank you very much. When connecting rotary push pin to GPIO25 anything is fine :)
謝謝你

I got the (wrong) info for GPIO27 from the schematic of this HW project https://github.com/stefschin/BrewPiLess-32-PCB.
In his schematic ROT_P is connected to GPIO27  :(